T-vK/ESP32-BLE-Keyboard #246: ESP32 connect to Android 1st time correctly, in 2nd connection Android forgets ESP32 and never reconnetting

lubaldi, 2023-08-07:
Hi, 
i've Esp32-c3-mini and and Android 12 smartphone.

in 1st connection all works well, but if i shut down bluetooth, or switch-off Esp32, once I turn it back on, Smartphone deletes the Esp32 from the list of Pairede device and does't view Esp32 in the list of avaiable device anymore.
i've to shut down bluetooth, or switch-off Esp32, then Smartphone sees it and reconnect how 1st time

Debugging i found that on the 2nd connection, Esp32 connects and immediatply disconnect.

i've this issue only on Android 12 (model Ulefone 12), not in other version of android

luissperber, 2023-08-27:
Are you using a NimBLE mode? (#247)

lubaldi, 2023-08-28:
of course

lubaldi, 2023-09-11:
i find this problem also on 
HugeRock X7 rugged tablet
Android 13