T-vK/ESP32-BLE-Keyboard #254: Send Regular Key Press

manek43509, 2023-09-21:
Hi, I'm sure this is a dumb question, but I still really need an answer!

I know I can use `bleKeyboard.press(KEY_RIGHT_ARROW);` to send a key press for a function key (in this case right arrow), but how can I use this library to send a single key press of a 'regular' letter key on the keyboard (for example, the letter 'L')? For some reason, I can't seem to find that information anywhere out there.

ekoslav, 2023-09-21:
bleKeyboard.write(32); - space
bleKeyboard.write(65);- letter A
bleKeyboard.write(76); - letter L

Use ASCII codes

manek43509, 2023-09-21:
Thanks, that's really helpful!

Probably another stupid question... but is there a full list of these codes for every key on a standard keyboard out there somewhere easily accessible? Just for reference, y'know?

DJprasenjit, 2024-01-18:
> Just search for "ASCII table". Or https://www.asciitable.com/
> 
> You can also test these codes let's say in Notepad using "ALT keys": Hold "ALT" key down and while holding it enter the code on the numeric part of the keyboard (for example 065) and the release the "ALT" key.

what is the code for back space (clear) and Caps lock??