T-vK/ESP32-BLE-Keyboard #255: No characters after reboot

mpeill, 2023-10-05:
Is there someting I'm missing here. THe ESP32-BLE-Keyboard works great when downloaded to the ESP32, then paired with a windows pc (or a tablet as well).
But after a power off/on of the ESP32, my code still runs, as shown by the flashing LED, but no characters are transmitted to the pc until I unpair, then pair at the pc again, even though the bluetooth on the pc shows the device being disconnected, then connected again at power on of the ESP32.
Here's my code:

#include <BleKeyboard.h>
BleKeyboard bleKeyboard("Page Turner","Malcolm Peill",100);

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleKeyboard.begin();
  }

void loop() {
  if(bleKeyboard.isConnected()) {
    Serial.println("Sending right arrow");
    bleKeyboard.write(KEY_RIGHT_ARROW);
    digitalWrite(LED_BUILTIN, HIGH);
    delay(2000);
    Serial.println("Sending left arrow");
    bleKeyboard.write(KEY_LEFT_ARROW);
    digitalWrite(LED_BUILTIN, LOW);
  }
  Serial.println("Waiting 5 seconds...");
  delay(5000);
}

raduprv, 2023-10-30:
I had the same problem and it got fixed after switching to Nimble. But I had to do a last unpair/pair. Now it seems to work fine.

mpeill, 2023-10-30:
Agreed, mine works fine too now, having fixed the other issue with the errors in the zip file that you helped solve.
Thanks again.