T-vK/ESP32-BLE-Keyboard #258: How to configure for NimBLE

mpeill, 2023-10-18:
I'm a relative Newbie so may be simple. I've uncommented the #define USE_NIMBLE at the start of BleKeyboard.h, and installed the NimBLE-Arduino library, but get errors when compiling, the first two being:

**In file included from D:\Malcolm\Documents\Arduino\libraries\ESP32-BLE-Keyboard\BleKeyboard.cpp:16:
D:\Malcolm\Documents\Arduino\libraries\ESP32-BLE-Keyboard\BleKeyboard.h:15:36: error: 'NimBLEHIDDevice' does not name a type; did you mean 'NimBLEDevice'?
 #define BLEHIDDevice               NimBLEHIDDevice
                                    ^~~~~~~~~~~~~~~
D:\Malcolm\Documents\Arduino\libraries\ESP32-BLE-Keyboard\BleKeyboard.h:110:3: note: in expansion of macro 'BLEHIDDevice'
   BLEHIDDevice* hid;
   ^~~~~~~~~~~~**

And more errors, but assuming they may all resolve after fixing the first, any thoughts on what I've missed?

raduprv, 2023-10-29:
I am having the same issue. No idea how to fix it..

raduprv, 2023-10-30:
Ok, so I found the fix! Do NOT use the release zip. Instead, just get the cpp and h file from github and use them instead.