T-vK/ESP32-BLE-Keyboard #264: Doesn't connect after the first time

facundosala, 2023-11-13:
Hello everyone. I want to clarify that I have uncommented the nidble line.
I have a successful connection the first time. I can send the commands I need.
When I disconnect because I move away from the coverage radius, it reconnects, but the commands I send it do not respond. The only solution is to forget and re-register the device from my Android and thus begin the cycle.
I cannot solve this problem or the solution you propose does not fix it. I have also used previous versions according to a comment that talks about the same thing.
Does anyone have a similar experience with this?
Thanks

turbo2ltr, 2023-11-16:
I'm experiencing the same issue, I am not using NimBLE.
Conneting to Android 12.  Connects the first time I pair.  I'm able to send data.  But when it disconnects for any reason, such as reflash a new sketch, or reboot the android device, it will not reconnect without deleting the device from the android device and re-pairing.

turbo2ltr, 2023-11-16:
I got this working better, but not perfect. In BLEKeyaboard.cpp change:
` pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND); `
to
` pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM); `

I tried a bunch of the other options and found it definitely doesn't  like bonding.  When I remove the bonding, the connection will survive a reboot of the android device.  I did not find any benefit to the less secure options. 

Unfortunately,  I have not been able to get it to survive a reboot of the ESP.  

Note this change is in the non NimBLE code, so you'll have to disable NimBLE for it to do anything.  I didn't look into NimBLE as it would not compile when I enabled it so I went back to not using it.  There is another security related function call for when  NimBLE is enabled. You may want to look there.

facundosala, 2023-11-21:
> I got this working better, but not perfect. In BLEKeyaboard.cpp change: `pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND);` to [...]

I don't understand if you were able to solve it in part. You say that you managed to get it to survive the restart and then you say that you didn't.

turbo2ltr, 2023-11-21:
Sorry. The using the original library, the connection would not survive a reboot of the **android device**.
After making the change, it would survive the reboot of the android device.

I have not been able to get it to survive a reboot of the ESP.

facundosala, 2023-11-22:
I found the solution within the forum. It's #174.
 I copy the solution:

Modify the BleKeyboard.cpp file and edit void BleKeyboard::end(void) { } with void BleKeyboard::end(void) { BLEDevice::deinit(); } Now you should be able to disconect ble. Right now, the program will auto reconnect so comment out advertising->start(); with //advertising->start(); in the function
void BleKeyboard::onDisconnect(BLEServer* pServer) {

To use this, you have to start advertizing the service with bleKeyboard.begin();. To stop BLE, run bleKeyboard.end();.
To restart BLE simply run bleKeyboard.begin(); again.

turbo2ltr, 2023-11-22:
That sounds like a solution to a different issue I was having where trying to disconnect from the android side doesn't seem to want to disconnect. But I will try anyway, thanks.

facundosala, 2023-11-22:
OKAY. let me know the result.

facundosala, 2023-11-23:
> That sounds like a solution to a different issue I was having where trying to disconnect from the android side doesn't seem to want to disconnect. But I will try anyway, thanks.

did you solve it?

distancerunner, 2023-11-28:
Now it works with my Android Device properly.

I renamed the keyboard name and removed it again from the android device.
Line 6 in the example file: BleKeyboard bleKeyboard("Fake Keyboard", "Funny Device Maker", 100);

Than I enabled nimble in the h file. Line 2

Compiled it again and testet it with connection to my pc -> was working

I think the removing from the android device must be done properly. Now it is reconnection as it should, every time after I upload a new code to the esp.

CNCmachineMaker, 2023-11-28:
Hi there,
I'm facing the same problem: no reconnect after reboot or disconnect from te ESP32 with windows10. 
I tried the above solution(s), unfortunately with no success.
Any suggestions?

distancerunner, 2023-11-29:
> Hi there, I'm facing the same problem: no reconnect after reboot or disconnect from te ESP32 with windows10. I tried the above solution(s), unfortunately with no success. Any suggestions?

Maybe try to compile with nimble = on and retry connection to another device. I think it is an issue, if you reconnect, without clearing bluetooth device properly from the machine.

Today it also works great only with nimble = on and on my android device.

CNCmachineMaker, 2023-11-29:
It is working!! SOmehoqw, I downloaded the wrong .h file. The #define USE_NIMBLE was missing.
After downloading the new file and uncommenting line 2 it works well.
Many thanks for your help!!

facundosala, 2023-11-30:
> It is working!! SOmehoqw, I downloaded the wrong .h file. The #define USE_NIMBLE was missing.
> After downloading the new file and uncommenting line 2 it works well.
> Many thanks for your help!!

What version did you use for the solution? Which downloader didn't work for you before?

CNCmachineMaker, 2023-11-30:
My apologies, I was a little premature in indicating that the problem had been resolved. The problem is still there.
There is another library on the gitHub: https://gist.github.com/manuelbl/66f059effc8a7be148adb1f104666467
reading the comments I noticed a remark that advertsing needs to be stopped before disconnect.
Unfortunately I am not an ecperienced programmer and do not have a clue what it means and what to change to get it working.

To answer your questions about versions, I copied the library properties. See below
I do hope a fix will come available.

Thanks for your info.

name=ESP32 BLE Keyboard
version=0.3.0
author=T-vK
maintainer=T-vK
sentence=Bluetooth LE Keyboard library for the ESP32.
paragraph=Bluetooth LE Keyboard library for the ESP32.
category=Communication
url=https://github.com/T-vK/ESP32-BLE-Keyboard
architectures=esp32

CNCmachineMaker, 2023-11-30:
> > Now it works with my Android Device properly. I had also the issue with reconnection. :(
> > I renamed the keyboard name and removed it again from the android device. Line 6 in the example file: BleKeyboard bleKeyboard("Fake Keyboard", "Funny Device Maker", 100);
> > Than I enabled nimble in the h file. Line 2
> > Compiled it again and testet it with connection to my pc -> was working
> > I think the removing from the android device must be done properly.
> > Now it is reconnecting as it should, every time after I upload a new code to the esp.
> 
> Could you specify well what you modified. I see that line 6 contains "BleKeyboard bleKeyboard;" What modification did you make?

I tried a different name after deleting the config in Windows: no success
Enabled Nimble by uncommeting line 2 in BleKeyboard.h : no success.

facundosala, 2023-12-01:
> My apologies, I was a little premature in indicating that the problem had been resolved. The problem is still there. There is another library on the gitHub: [...]

That's how it is. I spent several times thinking that it was solved and it is not.

It usually works the first few times and then it doesn't.

hravelo2020, 2024-05-11:
Hello, I have the same problem, I Can connect the first time, but when the connection is lost through the ESP32 or Android, which is my case, apparently it pairs according to Android, but it is not able to receive data from the ESP32, I think the The problem is in the void BleKeyboard::begin(void) function, which once initialized cannot restart any service. I tried to use the void BleKeyboard::end(void) command, but upon reviewing the .C file I see that it does nothing. The idea was to frequently ask from the ESP32 if the connection is alive, if not, disconnect using the BleKeyboard.End() command, but it does nothing within the BleKeyboard.c file, someone will know the instructions to generate this function to terminate the connection? In this way, in theory I should see the ESP32 as the first time we connect to it.

hravelo2020, 2024-05-11:
There are two problems, the first is when the Host is disconnected in my Android case, it could be because it is far away or out of range or because the tablet restarts, the second problem is when it restarts or something happens with the ESP32, in both cases once a previous successful connection is established, when retrying to connect a second time it fails, I am trying to understand the second case which is the restart of the ESP32, in this mode the bleKeyboard.begin function apparently tries to start a connection service that is already alive on the side of the computer or Host and that is where the error is generated. From the tests I have done, when the ESP32 is disconnected, it leaves several services or instances alive on the Host side. When the bleKeyboard.begin function is executed again, it does not check the existence of those instances. or live processes on the Hots side and that is where the problem is generated, I don't know what the process is or the connection mode, but I am investigating, if anyone has an idea I would be very grateful if you would evaluate the begin function, Regards.