T-vK/ESP32-BLE-Keyboard #270: does not compile with new esp32 core 3.0.0-alpha3

bsdinis1, 2023-12-08:
```
c:\Users\Admin\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp: In member function 'void BleKeyboard::begin()':
c:\Users\Admin\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:106:19: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
  106 |   BLEDevice::init(deviceName);
      |                   ^~~~~~~~~~
      |                   |
      |                   std::string {aka std::__cxx11::basic_string<char>}
In file included from c:\Users\Admin\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:9:
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLEDevice.h:42:40: note:   initializing argument 1 of 'static void BLEDevice::init(String)'
   42 |         static void        init(String deviceName);   // Initialize the local BLE environment.
      |                                 ~~~~~~~^~~~~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:117:32: error: no matching function for call to 'BLECharacteristic::setValue(std::string&)'
  117 |   hid->manufacturer()->setValue(deviceManufacturer);
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLEHIDDevice.h:17,
                 from c:\Users\Admin\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.h:24,
                 from c:\Users\Admin\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:1:
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:77:14: note: candidate: 'void BLECharacteristic::setValue(uint8_t*, size_t)'
   77 |         void setValue(uint8_t* data, size_t size);
      |              ^~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:77:14: note:   candidate expects 2 arguments, 1 provided
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:78:14: note: candidate: 'void BLECharacteristic::setValue(String)'
   78 |         void setValue(String value);
      |              ^~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:78:30: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
   78 |         void setValue(String value);
      |                       ~~~~~~~^~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:79:14: note: candidate: 'void BLECharacteristic::setValue(uint16_t&)'
   79 |         void setValue(uint16_t& data16);
      |              ^~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:79:33: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'uint16_t&' {aka 'short unsigned int&'}
   79 |         void setValue(uint16_t& data16);
      |                       ~~~~~~~~~~^~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:80:14: note: candidate: 'void BLECharacteristic::setValue(uint32_t&)'
   80 |         void setValue(uint32_t& data32);
      |              ^~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:80:33: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'uint32_t&' {aka 'long unsigned int&'}
   80 |         void setValue(uint32_t& data32);
      |                       ~~~~~~~~~~^~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:81:14: note: candidate: 'void BLECharacteristic::setValue(int&)'
   81 |         void setValue(int& data32);
      |              ^~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:81:28: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'int&'
   81 |         void setValue(int& data32);
      |                       ~~~~~^~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:82:14: note: candidate: 'void BLECharacteristic::setValue(float&)'
   82 |         void setValue(float& data32);
      |              ^~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:82:30: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'float&'
   82 |         void setValue(float& data32);
      |                       ~~~~~~~^~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:83:14: note: candidate: 'void BLECharacteristic::setValue(double&)'
   83 |         void setValue(double& data64);
      |              ^~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:83:31: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'double&'
   83 |         void setValue(double& data64);
      |                       ~~~~~~~~^~~~~~

exit status 1

Compilation error: exit status 1
`´´

Cflyyyy, 2023-12-18:
Have you made any progress with this? I went in and changed the default deviceNames values but that just made new issues show up.

bsdinis1, 2023-12-18:
you must use esp core 2.0.11 or esp core 2.0.10

Cflyyyy, 2024-01-14:
> you must use esp core 2.0.11 or esp core 2.0.10

Thanks so much!!! I thought I was going crazy.

2.0.11 seemed to worked better for me

SalamunKawlam, 2024-03-12:
> > you must use esp core 2.0.11 or esp core 2.0.10
> 
> Thanks so much!!! I thought I was going crazy.
> 
> 2.0.11 seemed to worked better for me

How did you get this version of esp core? I can only use the stable release (v2.0.14) or dev release (v3.0.0 alpha 3) from the additional board manager url.

valentingregoire, 2024-05-28:
Using version 2.0.11 actually worked! For people not knowing how to do this: you can select the version in the boards manager when installing esp32 by espressif:

[image]

airooter, 2024-06-12:
The error message indicates that there is a problem with converting a std::string to an Arduino String object. The code you provided tries to pass a std::string object as an argument to a function that expects an Arduino String. The Arduino String class and the std::string class are different and do not have implicit conversions between them.

To fix this issue, you need to convert the std::string object to an Arduino String object. You can do this by using the c_str() function of the std::string class to obtain a C-style string and then constructing an Arduino String object from it.

Here's an example of how you can modify the code:

**BLEDevice::init(String(deviceName.c_str()));**
By calling c_str() on the deviceName std::string object, you obtain a const char* that can be used to construct an Arduino String object.

Similarly, if you encounter the same issue when calling setValue() on the BLECharacteristic object, you can use the same approach to convert the std::string to an Arduino String:

**hid->manufacturer()->setValue(String(deviceManufacturer.c_str()));**

Make sure you include the necessary header files for std::string and Arduino String in your code.

eeerrr1, 2024-10-19:
非常感谢，我在编译SurfaceDial-esp32c3时遇到了相同问题