T-vK/ESP32-BLE-Keyboard #280: Delay in press() function

Spyby123, 2024-01-19:
I want to control 3-Axis CNC Machine over BLE with several Buttons. When i hold the Button down, the ESP should act like the asssigned Key gets pressed repeatedly, just like when you hold down a key on a normal keyboard.
I think the BleKeyboard.press('x') is the right function.
But as the CNC-programm running on my Laptop can't handle the inputs if they are to fast i need a delay between the keypresses.
How can i implement it, perhaps in the .h or .cpp file?
Sorry for asking, im a ESP32/Arduino beginner.

Andreas-ovs, 2025-02-05:
[image]

Andreas-ovs, 2025-02-05:
В место KEY_RETURN ставь свою ('x')