T-vK/ESP32-BLE-Keyboard #285: Does not work properly on ESP32-C3

maker114, 2024-03-01:
After connecting to the computer, it will keep disconnecting and reconnecting

sclea, 2024-03-01:
I confirm the same behevor. ESP32-C3 Super Mini board, win11.
Example sketch uploaded.
Device reconnects each 2-3 secs.
Sample keystrokes in the sketch are not printed in notepad in win neither KEY_MEDIA_PLAY_PAUSE have no effect.

sclea, 2024-03-01:
UPD: serial monitor output after pairing:

```
01:53:34.505 -> Sending Play/Pause media key...
01:53:35.500 -> Waiting 5 seconds...
01:53:40.504 -> Waiting 5 seconds...
01:53:45.507 -> Waiting 5 seconds...
01:53:46.341 -> E (37414) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
01:53:46.341 -> E (37414) BT_SMP: smp_derive_link_key_from_long_term_key failed
01:53:46.341 -> 
01:53:46.341 -> E (37419) BT_BTM: btm_proc_smp_cback received for unknown device
01:53:46.438 -> E (37532) BT_BTM: Device not found
01:53:46.438 -> 
01:53:50.510 -> Waiting 5 seconds...
01:53:55.515 -> Waiting 5 seconds...
01:54:00.519 -> Waiting 5 seconds...
```

and somewhere below:

```
01:57:02.254 -> 3fca0460: 0x3fca05a7 0x00000000 0x3fca046c 0xffffffff 0x3fca046c 0x3fca046c 0x00000000 0x3fca0480
01:57:02.254 -> 3fca0480: 0xffffffff 0x3fca0480 0x3fca0480 0x00000000 0x00000100 0x00000001 0xd900ffff 0x00000000
01:57:02.286 -> 3fca04a0: 0xb33fffff 0x00000000 0x72617453 0x676e6974 0x454c4220 0x726f7720 0x74696157 0x20676e69
01:57:02.286 -> 3fca04c0: 0x65732035 0x646e6f63 0x40c7281d 0xefcf6048 0x28a63051 0x44a01bbf 0x90459c53 0x9fcf0e00
01:57:02.383 -> 
01:57:02.383 -> 
01:57:02.383 -> 
01:57:02.383 -> ELF file SHA256: 06f8c4f7bce5adf0
01:57:02.383 -> 
01:57:02.383 -> Rebooting...
01:57:02.383 -> ESP-ROM:esp32c3-api1-20210207
01:57:02.383 -> Build:Feb  7 2021
01:57:02.383 -> rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
01:57:02.383 -> Saved PC:0x40381950
01:57:02.383 -> SPIWP:0xee
01:57:02.383 -> mode:DIO, clock div:1
01:57:02.383 -> load:0x3fcd5810,len:0x438
01:57:02.383 -> load:0x403cc710,len:0x918
01:57:02.383 -> load:0x403ce710,len:0x25f4
01:57:02.383 -> entry 0x403cc710
01:57:02.576 -> Starting BLE work!
01:57:02.833 -> Waiting 5 seconds...
01:57:07.837 -> Waiting 5 seconds...
```

maker114, 2024-03-02:
I solved the problem by changing 
**pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND);**
 to 
**pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);**
 at about line 130 of BleKeyboard.cpp，which looks like a security protocol sort of thing

sclea, 2024-03-02:
I tried this -- no changes. I have find another example, it sated working out of the box
https://gist.github.com/manuelbl/66f059effc8a7be148adb1f104666467

Will compare.

JJO998123, 2024-05-06:
I can confirm that I'm also having the same issue on an adafruit QTPY-ESP32-C3.
Code worked perfectly on an older ESP32-DevkKit. Constant disconnect and reconnect.

Maker114's line changed didn't work for me unfortunately.

ciccior2004, 2024-05-10:
> I can confirm that I'm also having the same issue on an adafruit QTPY-ESP32-C3. Code worked perfectly on an older ESP32-DevkKit but on the new (smaller) ESP32-C£ board, constant disconnect and reconnect.
> 
> The code is looping as expected, the board isn't restarting it's just dropping the bluetooth connection
> 
> Maker114's line changed didn't work for me unfortunately.

Delete paired keyboard and create a new pairing with keyboard. 
For me it works.

JJO998123, 2024-05-10:
Okay, put the change in, uploaded to the board, deleted it in the bluetooth settings in windows 11 and reconnected and can confirm it works.

Talismancer, 2024-07-11:
Thanks Maker114...that fixed it. I had to delete the device & re-pair it like others have.