T-vK/ESP32-BLE-Keyboard #287: HELP !!! Windows 10 (2024) Update Causes No HID Keyboard Drivers to be Loaded on connect

creepscompilation, 2024-03-22:
This ESP32-BLE Keyboard library has been working GREAT for over 1 year.

However, as of March 19th, 2024, I have tested up to (6) different Windows PC's and all 
of them will CONNECT without any issues, but fail to load any HID keyboard drivers.

This results in the ESP32 BLE Keyboard connecting, but failing to OUTPUT and keyboard codes.
The ESP32 BLE Keyboard stills works GREAT on Windows 8, 11, MacOS, iOS, and Android OS.

HELP !!! I need to find some solution to get Windows 10 to active the ESP32 BLE Keyboard
and load the HID keyboard drivers so I can send keyboard codes again.
[image]

creepscompilation, 2024-04-04:
I resolved this issue..
Basically I allowed Arduino IDE 2.0 to auto-update the libraries every time I used it..
Somewhere along the line, one of the libraries was updated which caused this issue..
I installed Arduino IDE 2.0 on a NEW PC and the problem disappeared..

Warning: Be very careful allowing Arduino IDE 2.0 to auto-update the libraries.
"If it ain't broke, don't fix it"

nlabrad, 2024-07-06:
And what libraries are you talking about? Would be helpful to know if someone else runs into this.

creepscompilation, 2024-07-08:
nlabrad, previously, I had my Arduino IDE setup to automatically update libraries, boards, and drivers, so I was never able to identify what update caused the issue. 

I literally had to install everything clean PC to get the code working again. I then made sure to disable the automatic updates.
Note: This BLE Keyboard was downloaded and installed on the clean PC and worked well again.