T-vK/ESP32-BLE-Keyboard #291: Lots of errors trying to compile latest version in IDE 2.0

sskindell, 2024-03-26:
Any ideas to fix?
Thank you,
Steve

c:\Users\sskin\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp: In member function 'void BleKeyboard::begin()':
c:\Users\sskin\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:106:19: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
  106 |   BLEDevice::init(deviceName);
      |                   ^~~~~~~~~~
      |                   |
      |                   std::string {aka std::__cxx11::basic_string<char>}
In file included from c:\Users\sskin\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:9:
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLEDevice.h:42:40: note:   initializing argument 1 of 'static void BLEDevice::init(String)'
   42 |         static void        init(String deviceName);   // Initialize the local BLE environment.
      |                                 ~~~~~~~^~~~~~~~~~
c:\Users\sskin\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:117:32: error: no matching function for call to 'BLECharacteristic::setValue(std::string&)'
  117 |   hid->manufacturer()->setValue(deviceManufacturer);
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLEHIDDevice.h:17,
                 from c:\Users\sskin\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.h:24,
                 from c:\Users\sskin\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:1:
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:77:14: note: candidate: 'void BLECharacteristic::setValue(uint8_t*, size_t)'
   77 |         void setValue(uint8_t* data, size_t size);
      |              ^~~~~~~~
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:77:14: note:   candidate expects 2 arguments, 1 provided
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:78:14: note: candidate: 'void BLECharacteristic::setValue(String)'
   78 |         void setValue(String value);
      |              ^~~~~~~~
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:78:30: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
   78 |         void setValue(String value);
      |                       ~~~~~~~^~~~~
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:79:14: note: candidate: 'void BLECharacteristic::setValue(uint16_t&)'
   79 |         void setValue(uint16_t& data16);
      |              ^~~~~~~~
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:79:33: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'uint16_t&' {aka 'short unsigned int&'}
   79 |         void setValue(uint16_t& data16);
      |                       ~~~~~~~~~~^~~~~~
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:80:14: note: candidate: 'void BLECharacteristic::setValue(uint32_t&)'
   80 |         void setValue(uint32_t& data32);
      |              ^~~~~~~~
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:80:33: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'uint32_t&' {aka 'long unsigned int&'}
   80 |         void setValue(uint32_t& data32);
      |                       ~~~~~~~~~~^~~~~~
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:81:14: note: candidate: 'void BLECharacteristic::setValue(int&)'
   81 |         void setValue(int& data32);
      |              ^~~~~~~~
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:81:28: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'int&'
   81 |         void setValue(int& data32);
      |                       ~~~~~^~~~~~
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:82:14: note: candidate: 'void BLECharacteristic::setValue(float&)'
   82 |         void setValue(float& data32);
      |              ^~~~~~~~
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:82:30: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'float&'
   82 |         void setValue(float& data32);
      |                       ~~~~~~~^~~~~~
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:83:14: note: candidate: 'void BLECharacteristic::setValue(double&)'
   83 |         void setValue(double& data64);
      |              ^~~~~~~~
C:\Users\sskin\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\src/BLECharacteristic.h:83:31: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'double&'
   83 |         void setValue(double& data64);

TURBULENTE, 2024-05-08:
Hello, 
The same problem happens when compiling.  Any solution around here?

qbalsdon, 2024-06-02:
Hey there, I fixed this by making some really small changes to the `BleKeyboard.cpp` file in `/Users/USERNAME/Documents/Arduino/libraries/ESP32_BLE_Keyboard/BleKeyboard.cpp`:

```
LINE 106: BLEDevice::init(deviceName.c_str());
...
LINE 117: hid->manufacturer()->setValue(deviceManufacturer.c_str()); 
```

Unfortunately there are other issues for me, but it got it compiling!

Draxo1, 2024-06-04:
> Hey there, I fixed this by making some really small changes to the `BleKeyboard.cpp` file in `/Users/USERNAME/Documents/Arduino/libraries/ESP32_BLE_Keyboard/BleKeyboard.cpp`:
>
> ```
> LINE 106: [...]

Thanks !!

Worked for me.

normaligator, 2024-07-20:
thank you mate, this worked for me!!

Hussain-making, 2024-07-28:
> thank you mate, this worked for me!!

can please tell what should I do exactly?
I mean what should I add or replace in blekeyboard.cpp file

icandothat2, 2025-01-24:
Same issue as 
https://github.com/T-vK/ESP32-BLE-Keyboard/issues/313

sachinthaHeshan, 2025-10-24:
> Hey there, I fixed this by making some really small changes to the `BleKeyboard.cpp` file in `/Users/USERNAME/Documents/Arduino/libraries/ESP32_BLE_Keyboard/BleKeyboard.cpp`:
>
> ```
> LINE 106: [...]

This is working for me Thanks