T-vK/ESP32-BLE-Keyboard #293: Doesnt print to text editor-HELP URGENT

SamihaFansur, 2024-04-12:
I'm unable to print to my notepad or any text editor but it prints to my serial monitor(while debugging), I'm using the latest BleKeyboard library and an esp32-s3-mini-1. Any idea why? I've tested it out on another esp32 as well.

JJO998123, 2024-05-10:
I'm having issues with my ESP32-c3 (Adafruit QTPY), when you look on your bluetooth devices do you get a permanent bluetooth connection or are you constantly reconnencting?

SamihaFansur, 2024-05-10:
It keeps reconnecting

JJO998123, 2024-05-10:
This issue has been solved by post #285 I have it working on my Adafruit QTPY ESP32-C3, you could try that solution and see if it works for you.

You will need to change a line of code in the library, upload to your board, remove your bluetooth device and repair. Worked for me. But mine is a C3 board not an S3 board....

SamihaFansur, 2024-05-10:
thank you, it works

JJO998123, 2024-05-10:
Not my solution, just putting it all together. Glad it works for you!