T-vK/ESP32-BLE-Keyboard #300: Adafruit QTPY ESP32-S3 - Constant reconnections and not sending keyboard messages.

JJO998123, 2024-05-06:
Hi

Wanted to start a new thread with the ESP32-S3 (Adafruit QTPY) which I purchased having tested code / hardware on ESP32 - Dev kit.

The ESP32-S3 constantly connects / disconnection on windows 11 Home PC. Does not send any text over Bluetooth Connection.

The Serial output of the code is appropriately responding so the device is still running and isn't resetting. This points to a non-stable connection over BLE.

Thanks for any help. Have tried using an older branch (as suggested in some of the comments) and have changed the auth line 130 as suggested by another comment branch.

JJO998123, 2024-05-10:
This is now resolved, line 130 in solution #295, deleted from the bluetooth in windows and it's all working now!

zhoubinx, 2024-06-27:
thanks，@JJO998123. I met the same problem, and resolved by your hint! Thanks again!