T-vK/ESP32-BLE-Keyboard #301: BleKeyboard not communicate after reset

PierreJom, 2024-05-11:
Hi, i want to create a Bluetooth macropad based on a ESP32, for that I use the BleKeyboard librarie and everything work well until I reset my ESP32 or unplug the power supply.

It say me that it is connected and send the keys but nothing happened on my computer and same on my phone. the only solution i found is to delete the connection to repair after.

I'm using a ESP32 Wroom 32 from DIY more and i'm on a Window 10.

there is somthing i forget or i don't understand

PierreJom, 2024-05-12:
After some research i found something in my Window OS, the device seems paired but is not and window get an error and can't find it (Metadata batching failure)

ALex-Pash, 2024-05-26:
Same problem...everythink works until reset/power off...Have you solve the problem? Any clue?

[win10 / nano esp32 / nextion]

> After some research i found something in my Window OS, the device seems paired but is not and window get an error and can't find it (Metadata batching failure)

ALex-Pash, 2024-05-26:
Hell Yeeeah. Problem fully disappear after reinstall library from 3.0 to 3.2 beta. Fix: Fixed https://github.com/T-vK/ESP32-BLE-Keyboard/issues/115 since v3.1

icandothat2, 2025-01-24:
This is resolved in ESP32-BLE-Keyboard v0.3.2-beta
https://github.com/T-vK/ESP32-BLE-Keyboard/releases/tag/0.3.2-beta

kdsl01, 2025-05-07:
> Hell Yeeeah. Problem fully disappear after reinstall library from 3.0 to 3.2 beta. Fix: Fixed https://github.com/T-vK/ESP32-BLE-Keyboard/issues/115 since v3.1

Hi. May I ask how are you able to make this code work on nano ESP32? I loaded the example code to my device. Device can be discovered by my computer but no input can be made.

icandothat2, 2025-05-07:
I was not using an Arduino Nano, I was using a XIAO ESP32 C6