T-vK/ESP32-BLE-Keyboard #305: Getting Exit Status 1 every time I run a code with this library or even the example for that matter - PLZ HELP!

DrkRCR, 2024-06-11:
Here are the error messages-
c:\Users\vedde\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp: In member function 'void BleKeyboard::begin()':
c:\Users\vedde\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:105:19: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
  105 |   BLEDevice::init(deviceName);
      |                   ^~~~~~~~~~
      |                   |
      |                   std::string {aka std::__cxx11::basic_string<char>}
In file included from c:\Users\vedde\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:7:
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLEDevice.h:41:27: note:   initializing argument 1 of 'static void BLEDevice::init(String)'
   41 |   static void init(String deviceName);                                                            // Initialize the local BLE environment.
      |                    ~~~~~~~^~~~~~~~~~
c:\Users\vedde\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:116:32: error: no matching function for call to 'BLECharacteristic::setValue(std::string&)'
  116 |   hid->manufacturer()->setValue(deviceManufacturer);
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLEServer.h:23,
                 from C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLEDevice.h:21:
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLECharacteristic.h:77:8: note: candidate: 'void BLECharacteristic::setValue(uint8_t*, size_t)'
   77 |   void setValue(uint8_t *data, size_t size);
      |        ^~~~~~~~
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLECharacteristic.h:77:8: note:   candidate expects 2 arguments, 1 provided
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLECharacteristic.h:78:8: note: candidate: 'void BLECharacteristic::setValue(String)'
   78 |   void setValue(String value);
      |        ^~~~~~~~
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLECharacteristic.h:78:24: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
   78 |   void setValue(String value);
      |                 ~~~~~~~^~~~~
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLECharacteristic.h:79:8: note: candidate: 'void BLECharacteristic::setValue(uint16_t&)'
   79 |   void setValue(uint16_t &data16);
      |        ^~~~~~~~
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLECharacteristic.h:79:27: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'uint16_t&' {aka 'short unsigned int&'}
   79 |   void setValue(uint16_t &data16);
      |                 ~~~~~~~~~~^~~~~~
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLECharacteristic.h:80:8: note: candidate: 'void BLECharacteristic::setValue(uint32_t&)'
   80 |   void setValue(uint32_t &data32);
      |        ^~~~~~~~
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLECharacteristic.h:80:27: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'uint32_t&' {aka 'long unsigned int&'}
   80 |   void setValue(uint32_t &data32);
      |                 ~~~~~~~~~~^~~~~~
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLECharacteristic.h:81:8: note: candidate: 'void BLECharacteristic::setValue(int&)'
   81 |   void setValue(int &data32);
      |        ^~~~~~~~
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLECharacteristic.h:81:22: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'int&'
   81 |   void setValue(int &data32);
      |                 ~~~~~^~~~~~
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLECharacteristic.h:82:8: note: candidate: 'void BLECharacteristic::setValue(float&)'
   82 |   void setValue(float &data32);
      |        ^~~~~~~~
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLECharacteristic.h:82:24: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'float&'
   82 |   void setValue(float &data32);
      |                 ~~~~~~~^~~~~~
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLECharacteristic.h:83:8: note: candidate: 'void BLECharacteristic::setValue(double&)'
   83 |   void setValue(double &data64);
      |        ^~~~~~~~
C:\Users\vedde\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\BLE\src/BLECharacteristic.h:83:25: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'double&'
   83 |   void setValue(double &data64);
      |                 ~~~~~~~~^~~~~~

exit status 1

Compilation error: exit status 1

mikk149c, 2024-06-17:
I'm having the exact same problem

saccoUnibg, 2024-06-19:
same issue.

dkyazzentwatwa, 2024-06-20:
Same issue as well...no idea how to fix but AI says to rework BleKeyboard.cpp lol :(

dkyazzentwatwa, 2024-06-20:
> same issue.

Fixed it by looking at a recent pull request:
https://github.com/T-vK/ESP32-BLE-Keyboard/pull/303

All you gotta do is change BleKeyboard.h/c where he made the changes! Works like a charm now.

saccocristian, 2024-06-20:
thanks! Will try later

FreemanCarvalho, 2024-07-19:
Hey guys, im super new to ESP32's, but im having the same issue.

dkyazzentwatwa when you say change the BleKeyboard.h/c what do you mean by that? Can you specify exactly where I should go to make those changes? If you could help me on this I would really appreciate that man.

TriggerHappy-NZ, 2024-09-18:
Same problem here, would appreciate if someone could share instructions on how to fix

fredmunyao1, 2024-10-07:
Thanks, Your Replies really helped me I was almost giving up!

cortex3, 2024-11-10:
fyi for anyone else: just doing a find and replace for std::string is not enough enough as there are other changes in line 100 necessary (see PR linked above).

Why is this not getting merged if a fix is known for this issue?