T-vK/ESP32-BLE-Keyboard #312: Getting Errors While Trying to run the code in Arduino IDE

MohammedAyanTheGreat, 2024-07-13:
When I click compile or upload I get the following error

c:\Users\user\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp: In member function 'void BleKeyboard::begin()':
c:\Users\user\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:106:19: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
  106 |   BLEDevice::init(deviceName);
      |                   ^~~~~~~~~~
      |                   |
      |                   std::string {aka std::__cxx11::basic_string<char>}
In file included from c:\Users\user\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:9:
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLEDevice.h:41:27: note:   initializing argument 1 of 'static void BLEDevice::init(String)'
   41 |   static void init(String deviceName);                                                            // Initialize the local BLE environment.
      |                    ~~~~~~~^~~~~~~~~~
c:\Users\user\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:117:32: error: no matching function for call to 'BLECharacteristic::setValue(std::string&)'
  117 |   hid->manufacturer()->setValue(deviceManufacturer);
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLEHIDDevice.h:17,
                 from c:\Users\user\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.h:24,
                 from c:\Users\user\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:1:
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:77:8: note: candidate: 'void BLECharacteristic::setValue(uint8_t*, size_t)'
   77 |   void setValue(uint8_t *data, size_t size);
      |        ^~~~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:77:8: note:   candidate expects 2 arguments, 1 provided
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:78:8: note: candidate: 'void BLECharacteristic::setValue(String)'
   78 |   void setValue(String value);
      |        ^~~~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:78:24: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
   78 |   void setValue(String value);
      |                 ~~~~~~~^~~~~
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:79:8: note: candidate: 'void BLECharacteristic::setValue(uint16_t&)'
   79 |   void setValue(uint16_t &data16);
      |        ^~~~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:79:27: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'uint16_t&' {aka 'short unsigned int&'}
   79 |   void setValue(uint16_t &data16);
      |                 ~~~~~~~~~~^~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:80:8: note: candidate: 'void BLECharacteristic::setValue(uint32_t&)'
   80 |   void setValue(uint32_t &data32);
      |        ^~~~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:80:27: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'uint32_t&' {aka 'long unsigned int&'}
   80 |   void setValue(uint32_t &data32);
      |                 ~~~~~~~~~~^~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:81:8: note: candidate: 'void BLECharacteristic::setValue(int&)'
   81 |   void setValue(int &data32);
      |        ^~~~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:81:22: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'int&'
   81 |   void setValue(int &data32);
      |                 ~~~~~^~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:82:8: note: candidate: 'void BLECharacteristic::setValue(float&)'
   82 |   void setValue(float &data32);
      |        ^~~~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:82:24: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'float&'
   82 |   void setValue(float &data32);
      |                 ~~~~~~~^~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:83:8: note: candidate: 'void BLECharacteristic::setValue(double&)'
   83 |   void setValue(double &data64);
      |        ^~~~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:83:25: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'double&'
   83 |   void setValue(double &data64);
      |                 ~~~~~~~~^~~~~~

exit status 1

Compilation error: exit status 1

palmtoy, 2024-07-14:
Fix compatibility with String for latest Arduino core
https://github.com/T-vK/ESP32-BLE-Keyboard/pull/303/commits/61a0002d409f33b4a7b574484173410f22e13554

Hope to help you.

aignemb, 2024-10-25:
> Fix compatibility with String for latest Arduino core [61a0002](https://github.com/T-vK/ESP32-BLE-Keyboard/commit/61a0002d409f33b4a7b574484173410f22e13554)
> 
> Hope to help you.

Thank you so much, I've spent all day trying to figure this out and that worked!

icandothat2, 2025-01-24:
Lines 105 to 116 should be changed

    //!!! THIS IS THE FIX ->add ".c_str()" after deviceName
    BLEDevice::init(deviceName.c_str());//-----------------------UPDATE THIS LINE
    BLEServer* pServer = BLEDevice::createServer();
    pServer->setCallbacks(this);

    hid = new BLEHIDDevice(pServer);
    inputKeyboard = hid->inputReport(KEYBOARD_ID); // <-- input REPORTID from report map
    outputKeyboard = hid->outputReport(KEYBOARD_ID);
    inputMediaKeys = hid->inputReport(MEDIA_KEYS_ID);

    outputKeyboard->setCallbacks(this);

    //!!! THIS IS THE FIX ->add ".c_str()" after deviceManufacturer

    hid->manufacturer()->setValue(deviceManufacturer.c_str());//-----------------------UPDATE THIS LINE