T-vK/ESP32-BLE-Keyboard #313: Can not build ESP32-BLE-Keyboard of example in window

DangChinh1496, 2024-07-14:
debug screen is:

C:\Users\GUEST01\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp: In member function 'void BleKeyboard::begin()':
C:\Users\GUEST01\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:106:19: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
  106 |   BLEDevice::init(deviceName);
      |                   ^~~~~~~~~~
      |                   |
      |                   std::string {aka std::__cxx11::basic_string<char>}
In file included from C:\Users\GUEST01\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:9:
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLEDevice.h:41:27: note:   initializing argument 1 of 'static void BLEDevice::init(String)'
   41 |   static void init(String deviceName);                                                            // Initialize the local BLE environment.
      |                    ~~~~~~~^~~~~~~~~~
C:\Users\GUEST01\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:117:32: error: no matching function for call to 'BLECharacteristic::setValue(std::string&)'
  117 |   hid->manufacturer()->setValue(deviceManufacturer);
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLEHIDDevice.h:17,
                 from C:\Users\GUEST01\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.h:24,
                 from C:\Users\GUEST01\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:1:
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:77:8: note: candidate: 'void BLECharacteristic::setValue(uint8_t*, size_t)'
   77 |   void setValue(uint8_t *data, size_t size);
      |        ^~~~~~~~
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:77:8: note:   candidate expects 2 arguments, 1 provided
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:78:8: note: candidate: 'void BLECharacteristic::setValue(String)'
   78 |   void setValue(String value);
      |        ^~~~~~~~
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:78:24: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
   78 |   void setValue(String value);
      |                 ~~~~~~~^~~~~
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:79:8: note: candidate: 'void BLECharacteristic::setValue(uint16_t&)'
   79 |   void setValue(uint16_t &data16);
      |        ^~~~~~~~
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:79:27: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'uint16_t&' {aka 'short unsigned int&'}
   79 |   void setValue(uint16_t &data16);
      |                 ~~~~~~~~~~^~~~~~
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:80:8: note: candidate: 'void BLECharacteristic::setValue(uint32_t&)'
   80 |   void setValue(uint32_t &data32);
      |        ^~~~~~~~
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:80:27: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'uint32_t&' {aka 'long unsigned int&'}
   80 |   void setValue(uint32_t &data32);
      |                 ~~~~~~~~~~^~~~~~
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:81:8: note: candidate: 'void BLECharacteristic::setValue(int&)'
   81 |   void setValue(int &data32);
      |        ^~~~~~~~
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:81:22: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'int&'
   81 |   void setValue(int &data32);
      |                 ~~~~~^~~~~~
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:82:8: note: candidate: 'void BLECharacteristic::setValue(float&)'
   82 |   void setValue(float &data32);
      |        ^~~~~~~~
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:82:24: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'float&'
   82 |   void setValue(float &data32);
      |                 ~~~~~~~^~~~~~
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:83:8: note: candidate: 'void BLECharacteristic::setValue(double&)'
   83 |   void setValue(double &data64);
      |        ^~~~~~~~
C:\Users\GUEST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLECharacteristic.h:83:25: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'double&'
   83 |   void setValue(double &data64);
      |                 ~~~~~~~~^~~~~~

exit status 1

Compilation error: exit status 1

santalex, 2024-07-24:
if you are using Arduino IDE, you can try platformio, it feels like Arduino IDE's problem.

eticaretmuhendisi, 2024-08-25:
Esp32's core files have been modified with 3.0.0, so there is an incompatibility. The workaround is to revert to version 2.0.17 until the library developers bring an update.

Zer0TheObserver, 2024-09-09:
That's all you need to do:
1, Open BleKeyBoard.ccp in the ESP32_BLE_Keyboard lib
2, change "hid->manufacturer()->setValue(deviceManufacturer);" TO "hid->manufacturer()->setValue(String(deviceManufacturer.c_str()));"

3, change "BLEDevice::init(deviceName);" TO "BLEDevice::init(String(deviceName.c_str()));"
Run it again.

dragon2777, 2024-11-23:
> That's all you need to do: 1, Open BleKeyBoard.ccp in the ESP32_BLE_Keyboard lib 2, change "hid->manufacturer()->setValue(deviceManufacturer);" TO [...]

amazing. thank you

icandothat2, 2025-01-23:
> That's all you need to do: 1, Open BleKeyBoard.ccp in the ESP32_BLE_Keyboard lib 2, change "hid->manufacturer()->setValue(deviceManufacturer);" TO [...]

Came here to day this. Lines 105 to 116 should be changed

>   BLEDevice::init(deviceName**.c_str()**);
>   BLEServer* pServer = BLEDevice::createServer();
>    pServer->setCallbacks(this);
>  
>    hid = new BLEHIDDevice(pServer);
>    inputKeyboard = hid->inputReport(KEYBOARD_ID);  // <-- input REPORTID from report map
>    outputKeyboard = hid->outputReport(KEYBOARD_ID);
>    inputMediaKeys = hid->inputReport(MEDIA_KEYS_ID);
>  
 >   outputKeyboard->setCallbacks(this);
>  
 >   hid->manufacturer()->setValue(deviceManufacturer**.c_str()**);