T-vK/ESP32-BLE-Keyboard #314: EP32-C3, doen't reconnect after rebood - Android

solilu7, 2024-08-02:
Hi, 
I've EP32-C3 boards, with some Andorid devices (10 12 and 13 version, all chinese brand, oukitel, ulefone, hugerock) i have this behavior:

I connect Esp32 board to smatphone searching a new device, it connects.
Then i switch-off and then switch-on (or simply reboot) Esp32 board, on the smatphone ESP-32 appears on "device cennected" section  just for an instant, then disappears (obviusly it is not connected).
if i want reconnect i have to repeat all procedure (searching ecc).

i use:
-ESP32 BLE Keyboard @ 0.3.2
-already modified row 130 of _BleKeyboard.cpp_ in _`pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);`
(also tried using NIMBLE, same behavior)

hilman2, 2024-10-27:
Did you find a solotion? I have the same problem with win10/11 (works with ios, android, chromeos, linux and my LG tv)

anlhse61557, 2024-12-15:
I got same problem, did you find a solution ?

messerffm, 2025-01-08:
Same here with iOS and macOS. Any ideas?

icandothat2, 2025-01-24:
Try  ESP32-BLE-Keyboard v0.3.2-beta
https://github.com/T-vK/ESP32-BLE-Keyboard/releases/tag/0.3.2-beta