T-vK/ESP32-BLE-Keyboard #316: NimBLE on arduino IDE

ramon-angosto, 2024-08-07:
I can't get this code to work with NimBLE in the arduino IDE

/Users/ramonangostoartigues/Documents/Arduino/libraries/ESP32_BLE_Keyboard/BleKeyboard.h:176:16: error: 'virtual void BleKeyboard::onConnect(NimBLEServer*)' marked 'override', but does not override
  176 |   virtual void onConnect(BLEServer* pServer) override;
      |                ^~~~~~~~~
/Users/ramonangostoartigues/Documents/Arduino/libraries/ESP32_BLE_Keyboard/BleKeyboard.h:177:16: error: 'virtual void BleKeyboard::onDisconnect(NimBLEServer*)' marked 'override', but does not override
  177 |   virtual void onDisconnect(BLEServer* pServer) override;
      |                ^~~~~~~~~~~~
/Users/ramonangostoartigues/Documents/Arduino/libraries/ESP32_BLE_Keyboard/BleKeyboard.h:178:16: error: 'virtual void BleKeyboard::onWrite(NimBLECharacteristic*)' marked 'override', but does not override
  178 |   virtual void onWrite(NimBLECharacteristic* me) override;
      |                ^~~~~~~

exit status 1

Compilation error: exit status 1

OPiElix, 2024-12-03:
I got the same problem here.

Sevan05, 2024-12-11:
You need use NimBLE-arduino 1.4.3

awong1900, 2024-12-26:
> You need use NimBLE-arduino 1.4.3

Yes, break update from 1.x -> 2.x, https://github.com/h2zero/NimBLE-Arduino/blob/master/docs/1.x_to2.x_migration_guide.md 

So need update this lib for 2.x

Antonio65-esp, 2025-01-03:
@ramon-angosto conseguiste resolverlo? Yo estoy atascado en cuanto uso la librería BleKeyboard.h