T-vK/ESP32-BLE-Keyboard #320: Arduino Compile Error

hasantahsindagli, 2024-08-21:
Hi,
I wrote a code to send keyboard keys to pc with buttons using ESP32. But I am getting errors during compilation. I would be glad if you could help me.

[Compilation Error.txt](https://github.com/user-attachments/files/16690420/Compilation.Error.txt)
[ESP32.ino.txt](https://github.com/user-attachments/files/16690421/ESP32.ino.txt)

eticaretmuhendisi, 2024-08-25:
Esp32's core files have been modified with 3.0.0, so there is an incompatibility. The workaround is to revert to version 2.0.17 until the library developers bring an update.

https://forum.arduino.cc/t/ble-combo-failing-to-compile/1274352/2

hasantahsindagli, 2024-08-25:
> Hi @hasantahsindagli, Esp32's core files have been modified with 3.0.0, so there is an incompatibility. The workaround is to revert to version 2.0.17 until the library developers bring an update.
> 
> https://forum.arduino.cc/t/ble-combo-failing-to-compile/1274352/2

Teşekkür ederim

Halphas, 2024-10-16:
tenks brov