T-vK/ESP32-BLE-Keyboard #323: Have to re pair for my code to work

Volanaro, 2024-09-27:
what the title says, with my code:
if i pair the esp32 with my pc (win 11) it works fine, if i then pull the power out of the esp32 and re connect the power, it still connects to windows as it should but it dose not send the keys when the buttons are pressed?, not sure if this is a libary issue or my code issue, iv looked all over the internet and even asked chat gpt for help and im coming up with nothing?

`#include <BleKeyboard.h>

// Pin definitions (ensure these match your setup)
const int BUTTON_o_PIN = 12;    // Pin for Button L
const int BUTTON_p_PIN = 13;    // Pin for Button P

// Create a BleKeyboard object with String parameters
BleKeyboard bleKeyboard("YourDeviceName", "YourManufacturer"); // Pass strings directly

void setup() {
    // Initialize Serial for debugging
    Serial.begin(115200);
    
    // Initialize the BLE keyboard
    bleKeyboard.begin();

    // Set button pins as input with pull-up resistor
    pinMode(BUTTON_o_PIN, INPUT_PULLUP);
    pinMode(BUTTON_p_PIN, INPUT_PULLUP);
}

void loop() {
    // Read the current state of the buttons
    bool currentButtonLState = digitalRead(BUTTON_o_PIN) == LOW;  // Button pressed state
    bool currentButtonPState = digitalRead(BUTTON_p_PIN) == LOW;  // Button pressed state

    // Button L logic
    if (currentButtonLState) {
        bleKeyboard.press('o'); // Send key press for 'o'
        Serial.println("Button L Pressed");
        delay(100); // debounce delay
    } else {
        bleKeyboard.release('o'); // Release key for 'o'
    }

    // Button P logic
    if (currentButtonPState) {
        bleKeyboard.press('p'); // Send key press for 'p'
        Serial.println("Button P Pressed");
        delay(100); // debounce delay
    } else {
        bleKeyboard.release('p'); // Release key for 'p'
    }

    // Optional: Small delay to mitigate excessive keypresses
    delay(10);
}`

hilman2, 2024-10-27:
Did you find a solotion? I have the same problem with win10/11 (works with ios, android, chromeos, linux and my LG tv)

icandothat2, 2025-01-24:
This same problem is being discussed in this thread. 
https://github.com/T-vK/ESP32-BLE-Keyboard/issues/328

icandothat2, 2025-01-24:
This is resolved in ESP32-BLE-Keyboard v0.3.2-beta
https://github.com/T-vK/ESP32-BLE-Keyboard/releases/tag/0.3.2-beta