T-vK/ESP32-BLE-Keyboard #328: stops working after reconnect (windows only)

hilman2, 2024-10-27:
At first I thought it was due to my code, so I tested it with an absolute minimal configuration: same result.
Setup: Seed Studio XIAO ESP32C3 with button connected to D0<->GND
Arduino IDE with esp32 2.0.17
On Windows 10/11, the device is detected and can be paired.
After pairing, everything works perfectly. When you press the button, "i" characters are typed into Notepad.
When you restart the ESP, the Windows laptop, or even just the Bluetooth on the laptop, the paired device reconnects, but no letters are typed into Notepad anymore. Something is happening though, as the cursor briefly stops blinking.

This issue doesn't occur on iOS, Android, ChromeOS, or Linux.
I even tested it with a live system on the same hardware where it wasn't working under Windows:
With Ubuntu, there's no problem at all - after resetting the ESP, it reconnects immediately and you can type letters again.
If I remove the device under Windows and pair it again, it works again - until the next reconnection.
Any ideas? Or has anyone experienced the same problem?

The Code I use for testing.

```
#include <BleKeyboard.h>

// Initialize the BLE Keyboard with a name and manufacturer
BleKeyboard bleKeyboard("Button Keyboard", "MyCompany", 100);

// Define the button pin
const int BUTTON_PIN = 2; // D0 = GPIO2

// Variables for button debouncing
bool lastButtonState = HIGH; // Initial state (Pull-Up)
unsigned long lastDebounceTime = 0;
const unsigned long debounceDelay = 50; // Debounce time in milliseconds

void setup() {
  // Start serial communication for debugging (optional)
  Serial.begin(115200);
  Serial.println("Starting BLE Keyboard...");

  // Configure the button pin as input with internal pull-up resistor
  pinMode(BUTTON_PIN, INPUT_PULLUP);
  Serial.println("Button pin configured as INPUT_PULLUP.");

  // Start the BLE Keyboard
  bleKeyboard.begin();
  Serial.println("BLE Keyboard initialized.");
}

void loop() {
  // Check if the BLE Keyboard is connected
  if (bleKeyboard.isConnected()) {
    // Read the current state of the button
    int reading = digitalRead(BUTTON_PIN);

    // Check if the button state has changed (debouncing)
    if (reading != lastButtonState) {
      lastDebounceTime = millis();
    }

    // If the state has been stable for longer than the debounce delay
    if ((millis() - lastDebounceTime) > debounceDelay) {
      // If the button is pressed (LOW because of Pull-Up)
      if (reading == LOW) {
        Serial.println("Button pressed. Sending 'i'.");
        bleKeyboard.print("i");
        // Short delay to prevent multiple sends
        delay(200);
      }
    }

    // Update the last button state
    lastButtonState = reading;
  }

  // Short delay to reduce CPU usage
  delay(10);
}
```

Juzz-Layman, 2024-12-24:
i have the same issue. looking into a fix.

Justus38, 2024-12-25:
any solution for this problem? i also have it on windows and android

messerffm, 2025-01-08:
Same here with iOS and macOS. Any solution?

Alberto-D, 2025-01-22:
The same thing happens wiht both of my androids, after rebooting the esp32 it says it is conected in the phone but doesnt work, i have to manually unpair and pair again and then it works.

icandothat2, 2025-01-24:
I also have the same problem. If I remove the device from the blue tooth settings of my computer, then re-add the device while it's running, it will play/pause music and type works. If I then reset or power off and back on the chip (Xiao ESP32c6) the device is reconnected in the bluetooth panel in windows but the play/pause doesn't work until I delete it from bluetooth settings and then reconnect.

icandothat2, 2025-01-24:
This is resolved in ESP32-BLE-Keyboard v0.3.2-beta
https://github.com/T-vK/ESP32-BLE-Keyboard/releases/tag/0.3.2-beta

neiloe8023, 2025-02-22:
> 此问题已在 ESP32-BLE-Keyboard v0.3.2-beta https://github.com/T-vK/ESP32-BLE-Keyboard/releases/tag/0.3.2-beta 中得到解决
我在Windows11上进行了测试，仍然不行。在IOS上可以使用。Win11中一直在断开、重连。期待您的回复，谢谢

neiloe8023, 2025-02-22:
> This is resolved in ESP32-BLE-Keyboard v0.3.2-beta https://github.com/T-vK/ESP32-BLE-Keyboard/releases/tag/0.3.2-beta

I tested it on Windows 11, but it still doesn't work. It works on IOS. It keeps disconnecting and reconnecting in Win 11. Looking forward to your reply, thank you

hcabel, 2025-02-27:
Hey, I'm having the same issue on a totally different code base 
I'll be keen to hear what the fix was for you, I've look through the latest commit but could find anything related 🤔