T-vK/ESP32-BLE-Keyboard #330: Connection to Windows 11 fails

berteding, 2024-11-09:
Windows 11 sees me c32 s3 board but the connection toggles fast between  connected and disconnected.
Tried different usb device in my laptop, intel build in and realtek 5.4 Bluetooth usb stick. 
On Android phone an tablet no problem.
I disconnected my phone and tablet first before trying my Windows 11 laptop.

NopeProsOnly, 2024-11-10:
I'm able to reproduce this on my end as well. It worked for a bit, but suddenly, unexpectedly, it toggled the connection very quickly.

beding, 2024-11-11:
I found another BleKeyboard code here: https://gist.github.com/manuelbl/66f059effc8a7be148adb1f104666467
This gives a stable connection on Windows 11
But there is on one interface function called typeText(), so no keypress and release functions.
So i hope that T-Vk can compare this code with his and find the problem.

bligede767, 2024-12-26:
> I found another BleKeyboard code here: https://gist.github.com/manuelbl/66f059effc8a7be148adb1f104666467 This gives a stable connection on Windows 11 But there is only one interface function called typeText(), so no keypress and release functions. So i hope that T-Vk can compare this code with his and find the problem.

How can I implemnt the code? I am newbie. Please, help

berteding, 2024-12-26:
I can send you my project if you like, but then i need your email.
I implemented a macro keyboard for VLC player with the code of manuelbl, this works well on windows 11.
The ESP-32-BLE-Keyboard does not work with windows 11 and it looks as if nobody is going to solve it.

berteding, 2024-12-27:
Just add the code to your project, include the header file and use the functions as in the example.