T-vK/ESP32-BLE-Keyboard #340: Cannot understand reasons of next errors:

SergeZon, 2024-12-29:
In file included from D:\Arduino_Projects\RGT_BT_steering\RGT_BT_steering.ino:22:
d:\Users\Serge\Documents\Arduino\libraries\ESP32_BLE_Keyboard/BleKeyboard.h:131:60: error: expected class-name before ',' token
  131 | class BleKeyboard : public Print, public BLEServerCallbacks, public BLECharacteristicCallbacks
      |                                                            ^
d:\Users\Serge\Documents\Arduino\libraries\ESP32_BLE_Keyboard/BleKeyboard.h:19:36: error: 'NimBLEAdvertising' does not name a type; did you mean 'BLEAdvertising'?
   19 | #define BLEAdvertising             NimBLEAdvertising
      |                                    ^~~~~~~~~~~~~~~~~
d:\Users\Serge\Documents\Arduino\libraries\ESP32_BLE_Keyboard/BleKeyboard.h:138:3: note: in expansion of macro 'BLEAdvertising'
  138 |   BLEAdvertising*    advertising;
      |   ^~~~~~~~~~~~~~
d:\Users\Serge\Documents\Arduino\libraries\ESP32_BLE_Keyboard/BleKeyboard.h:176:16: error: 'virtual void BleKeyboard::onConnect(NimBLEServer*)' marked 'override', but does not override
  176 |   virtual void onConnect(BLEServer* pServer) override;
      |                ^~~~~~~~~
d:\Users\Serge\Documents\Arduino\libraries\ESP32_BLE_Keyboard/BleKeyboard.h:177:16: error: 'virtual void BleKeyboard::onDisconnect(NimBLEServer*)' marked 'override', but does not override
  177 |   virtual void onDisconnect(BLEServer* pServer) override;
      |                ^~~~~~~~~~~~
d:\Users\Serge\Documents\Arduino\libraries\ESP32_BLE_Keyboard/BleKeyboard.h:178:16: error: 'virtual void BleKeyboard::onWrite(NimBLECharacteristic*)' marked 'override', but does not override
  178 |   virtual void onWrite(BLECharacteristic* me) override;
      |                ^~~~~~~

StevT7, 2025-01-01:
The NimBLE library's API has been updated, necessitating several modifications to both the .h and .cpp files to align with the new interface.

StevT7, 2025-01-01:
BleKeyboard.h:

Include additional libraries
IS:

```
#include "NimBLECharacteristic.h"
#include "NimBLEHIDDevice.h"
```

NEW:
```
#include "NimBLECharacteristic.h"
#include "NimBLEHIDDevice.h"
#include "NimBLEAdvertising.h"
#include "NimBLEServer.h"
```

Change the following lines

IS:

```
virtual void onConnect(BLEServer* pServer) override;
  virtual void onDisconnect(BLEServer* pServer) override;
  virtual void onWrite(BLECharacteristic* me) override;
```

NEW:

```
virtual void onConnect(BLEServer* pServer, NimBLEConnInfo & connInfo) override;
  virtual void onDisconnect(BLEServer* pServer, NimBLEConnInfo & connInfo, int reason) override;
  virtual void onWrite(BLECharacteristic* me, NimBLEConnInfo & connInfo) override;
```

byerss, 2025-01-03:
I've temporality worked around the issue by using the old version of NimBLE in PlatformIO like so:

```
lib_deps = 
    T-vk/ESP32 BLE Keyboard@^0.3.2-beta
    NimBLE-Arduino@^1.4.3
build_flags = 
    -D USE_NIMBLE
```