T-vK/ESP32-BLE-Keyboard #347: errors converting std::string to String

haywireSSC, 2025-01-24:
[image]
on line 117 and 106 throws errors unless explicitly cast from string to c string. I added .c_str to each now it works

TriDEntApollO, 2025-01-27:
Refer to this pull request #343  of mine for the fix

CodeFoxy-Github, 2025-02-15:
1. **Convert `std::string` to `String` in `BleKeyboard.cpp`:**
   - Locate `BLEDevice::init(deviceName);` and change it to:  
     ```cpp
     BLEDevice::init(String(deviceName.c_str()));
     ```
   - Locate `hid->manufacturer()->setValue(deviceManufacturer);` and change it to:  
     ```cpp
     hid->manufacturer()->setValue(String(deviceManufacturer.c_str()));
     ```