T-vK/ESP32-BLE-Keyboard #349: Problems with android linking.

Marceloktm2, 2025-02-05:
Hello guys, I'm new here. I wanted to tell you about a problem that is giving me a headache. 
It mainly focuses on linking with any Android device. 
The device appears visible but when it comes to linking, an error appears that does not allow connection. 

I am specifically using an esp32 S3 Zero with the latest Blekeyboard library 0.3.2

Let's see if the same thing happens to anyone or has a solution because I'm stuck.
I have also changed lines 106 and 117 of the library with [.c_str()]

 This is the code I´m using:

#include <Arduino.h>
#include <BleKeyboard.h>

BleKeyboard bleKeyboard("ESP32_Keyboard", "ESP32_Dev", 100);
const int buttonPin = 9;  // Cambia al GPIO de tu botón

void restartBLE() {
  Serial.println("Reiniciando BLE para asegurar reconexión...");
  bleKeyboard.end();
  delay(500);
  bleKeyboard.begin();
}

void setup() {
  Serial.begin(115200);
  pinMode(buttonPin, INPUT_PULLUP);  

  Serial.println("Iniciando BLE Keyboard...");
  bleKeyboard.begin();
  
  unsigned long startTime = millis();
  while (!bleKeyboard.isConnected() && millis() - startTime < 5000) {  
    Serial.println("Esperando conexión BLE...");
    delay(500);
  }

  if (bleKeyboard.isConnected()) {
    Serial.println("BLE Conectado!");
    delay(1000);  
    bleKeyboard.write(KEY_MEDIA_VOLUME_UP);
    Serial.println("Primer comando enviado para activar HID.");
  } else {
    Serial.println("BLE no conectado, intentando reiniciar...");
    restartBLE();
  }
}

void loop() {
  if (digitalRead(buttonPin) == LOW) {  
    Serial.println("Botón presionado, enviando volumen arriba...");
    if (bleKeyboard.isConnected()) {
      bleKeyboard.write(KEY_MEDIA_VOLUME_UP);
    } else {
      Serial.println("BLE desconectado, intentando reiniciar...");
      restartBLE();
    }
    delay(300);
  }
}

Marceloktm2, 2025-02-05:
Investigating I have found what the problem seems to be. It turns out that it gave errors when pairing using the link security codes. 
What I have done is basically override it from the code.
I would like to preserve that security but in the meantime it is functional. I hope it helps you and a solution is found soon.

I leave you the code that needs to be added:

```
#include <BLEDevice.h>
#include <BLESecurity.h>

BLESecurity *pSecurity = new BLESecurity();

void setup() {

  BLEDevice::init("DEVICENAME");
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_NO_BOND);
  pSecurity->setCapability(ESP_IO_CAP_NONE);
  pSecurity->setInitEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);

  bleKeyboard.begin();
}

void loop() {

}
```

paloky, 2025-02-07:
Hi Marceloktm2.
I have the same problem and with this works fine with version 0.3.2-beta !!.

Muchas gracias.
Thank's a lot !!

pablomuri, 2025-05-27:
It works like a charm