T-vK/ESP32-BLE-Keyboard #354: trying to build a page turner with esp32, Compilation error: exit status 1

stevelovessun, 2025-03-16:
Hi all

I hope this is put into the correct place .
first post be kind.

I am trying to build a page turner with esp32
https://codeandlife.com/2021/05/23/diy-bluetooth-sheet-music-page-turn-pedal-with-esp32/

The code i am using:

#include <BleKeyboard.h>

// shorter 15 character name seems to work better w. iPad
BleKeyboard bleKeyboard("ESP32-Keyboard2", "Me", 100);

#define BUTTON_PIN 4

void setup() {
  pinMode(BUTTON_PIN, INPUT_PULLUP);
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleKeyboard.begin();
}

void loop() {
  static int debug = 0, counter = 0;
  char button = digitalRead(BUTTON_PIN) == LOW;

  if(button) counter++;
  
  if(bleKeyboard.isConnected() && !button && counter > 1) {
    Serial.println(counter);
   
    bleKeyboard.write(counter < 10 ? KEY_RIGHT_ARROW : KEY_LEFT_ARROW);
    delay(100);
    bleKeyboard.releaseAll();

    counter = 0;
  }

  if(++debug % 20 == 0) Serial.println("sec");
  delay(50);
}

but i am getting the following errors:

C:\Users\steve\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp: In member function 'void BleKeyboard::begin()':
C:\Users\steve\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:105:19: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
  105 |   BLEDevice::init(deviceName);
      |                   ^~~~~~~~~~
      |                   |
      |                   std::string {aka std::__cxx11::basic_string<char>}
In file included from C:\Users\steve\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:7:
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLEDevice.h:41:27: note:   initializing argument 1 of 'static void BLEDevice::init(String)'
   41 |   static void init(String deviceName);                                                            // Initialize the local BLE environment.
      |                    ~~~~~~~^~~~~~~~~~
C:\Users\steve\Documents\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:116:32: error: no matching function for call to 'BLECharacteristic::setValue(std::string&)'
  116 |   hid->manufacturer()->setValue(deviceManufacturer);
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLEServer.h:23,
                 from C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLEDevice.h:21:
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLECharacteristic.h:83:8: note: candidate: 'void BLECharacteristic::setValue(double&)'
   83 |   void setValue(double &data64);
      |        ^~~~~~~~
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLECharacteristic.h:83:25: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'double&'
   83 |   void setValue(double &data64);
      |                 ~~~~~~~~^~~~~~
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLECharacteristic.h:82:8: note: candidate: 'void BLECharacteristic::setValue(float&)'
   82 |   void setValue(float &data32);
      |        ^~~~~~~~
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLECharacteristic.h:82:24: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'float&'
   82 |   void setValue(float &data32);
      |                 ~~~~~~~^~~~~~
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLECharacteristic.h:81:8: note: candidate: 'void BLECharacteristic::setValue(int&)'
   81 |   void setValue(int &data32);
      |        ^~~~~~~~
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLECharacteristic.h:81:22: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'int&'
   81 |   void setValue(int &data32);
      |                 ~~~~~^~~~~~
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLECharacteristic.h:80:8: note: candidate: 'void BLECharacteristic::setValue(uint32_t&)'
   80 |   void setValue(uint32_t &data32);
      |        ^~~~~~~~
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLECharacteristic.h:80:27: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'uint32_t&' {aka 'long unsigned int&'}
   80 |   void setValue(uint32_t &data32);
      |                 ~~~~~~~~~~^~~~~~
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLECharacteristic.h:79:8: note: candidate: 'void BLECharacteristic::setValue(uint16_t&)'
   79 |   void setValue(uint16_t &data16);
      |        ^~~~~~~~
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLECharacteristic.h:79:27: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'uint16_t&' {aka 'short unsigned int&'}
   79 |   void setValue(uint16_t &data16);
      |                 ~~~~~~~~~~^~~~~~
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLECharacteristic.h:77:8: note: candidate: 'void BLECharacteristic::setValue(uint8_t*, size_t)'
   77 |   void setValue(uint8_t *data, size_t size);
      |        ^~~~~~~~
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLECharacteristic.h:77:8: note:   candidate expects 2 arguments, 1 provided
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLECharacteristic.h:78:8: note: candidate: 'void BLECharacteristic::setValue(String)'
   78 |   void setValue(String value);
      |        ^~~~~~~~
C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLECharacteristic.h:78:24: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
   78 |   void setValue(String value);
      |                 ~~~~~~~^~~~~

exit status 1

Compilation error: exit status 1

I hope this is a simple fix.
all i have programmed was wled which was an online install, so easy .

Many thanks

ekoslav, 2025-03-16:
This may have been in regards of using "std::string" vs "String". There is an open issue (https://github.com/T-vK/ESP32-BLE-Keyboard/issues/353#issuecomment-2724594844), and if you do not use NIMBle, this might work for you:
https://github.com/T-vK/ESP32-BLE-Keyboard/issues/353#issuecomment-2724594844
Replace BleKeyboard.h and BleKeyboard.cpp , and see if it compiles.

stevelovessun, 2025-03-17:
> This may have been in regards of using "std::string" vs "String". There is an open issue ([#353 (comment)](https://github.com/T-vK/ESP32-BLE-Keyboard/issues/353#issuecomment-2724594844)), and if you do not use NIMBle, this might work for you: [#353 (comment)](https://github.com/T-vK/ESP32-BLE-Keyboard/issues/353#issuecomment-2724594844) Replace BleKeyboard.h and BleKeyboard.cpp , and see if it compiles.

Thank You this fix the problem!

ekoslav, 2025-03-17:
This is thanks to Norbert!

stevelovessun, 2025-03-17:
one last thing

with the following code:

what do i need to change?
as all i want is 1 button to pause video on one press and then rewind 10 -15 sec on long press?

#include <BleKeyboard.h>

// shorter 15 character name seems to work better w. iPad
BleKeyboard bleKeyboard("ESP32-Keyboard2", "Me", 100);

#define BUTTON_PIN 4

void setup() {
pinMode(BUTTON_PIN, INPUT_PULLUP);
Serial.begin(115200);
Serial.println("Starting BLE work!");
bleKeyboard.begin();
}

void loop() {
static int debug = 0, counter = 0;
char button = digitalRead(BUTTON_PIN) == LOW;

if(button) counter++;

if(bleKeyboard.isConnected() && !button && counter > 1) {
Serial.println(counter);

bleKeyboard.write(counter < 10 ? KEY_RIGHT_ARROW : KEY_LEFT_ARROW);
delay(100);
bleKeyboard.releaseAll();

counter = 0;
}

if(++debug % 20 == 0) Serial.println("sec");
delay(50);
}

I hope someone can writ in the correct code for me please?

I understand i need KEY_MEDIA_PLAY_PAUSE but not sure the code to rewind 10-15 seconds backwards?

Many thanks