T-vK/ESP32-BLE-Keyboard #357: Seeed ESP32 C3: Connecting, but not sending?

chrisallick, 2025-03-21:
I started working with the Arduino Nano ESP32 and was able to get the library working, connecting, and sending keystrokes.

I was really excited to work with this: https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html because this board has a batter charging circuit. So it'll charge off usbc, and then switch to battery power.

However, after getting it to compile by using a branched version, I am not seeing any data come through after connecting to computer.

I did some error messages here and there, but was unclear if they were significant or not. Again, the Arduino Nano ESP32 worked exactly as expected, just ran the example and connected on my other Mac OS device, data started coming through.

Any thoughts? Or... any go-to options for a bluetooth mcu, hid, low power? the Seeed ESP32 C3 boasts the battery management and deepsleep options. Unfortunate it's not working, hoping it's a simple config option or something trivial.

```
E (114648) BT_BTM: BTM_GetSecurityFlags false
E (114648) BT_GATT: GATT_INSUF_AUTHENTICATION
E (116538) BT_BTM: BTM_GetSecurityFlags false
E (114108) BT_GATT: GATT_INSUF_AUTHENTICATION
E (114439) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
E (114439) BT_SMP: smp_derive_link_key_from_long_term_key failed
```

HubKing, 2025-03-25:
I have a cheap no-brand C3, and when I ran the example code, there was an error after the pairing confirmation dialogue, so the keyboard was not connected. [A code from another person ](https://gist.github.com/manuelbl/66f059effc8a7be148adb1f104666467?permalink_comment_id=4725735)worked, but the code was not using any library, so the code was complex.

chrisallick, 2025-03-25:
Ah, ok. Well I got it working! I switched to the XIAO ESP32 S3! This board is the same small size, has a battery charger, and works while plugged in. So you can charge while tethered via USB C.

I applied two changes in BleKeyboard.cpp

First is the compile error fix:

```
void BleKeyboard::begin(void)
{
  BLEDevice::init(deviceName.c_str()); // <-- chris allick https://forum.arduino.cc/t/help-with-compile-error/1283905/5
```

and

`hid->manufacturer()->setValue(deviceManufacturer.c_str());  // <-- chris allick`

I also, changed the following

```
  BLESecurity* pSecurity = new BLESecurity();
  //pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND); // <- chris allick
  // https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_security_server/tutorial/Gatt_Security_Server_Example_Walkthrough.md
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);
```

I tried all the AuthenticationMode options and ESP_LE_AUTH_BOND worked! I have to do more research into these modes and the implications, but, for my use which is a simple at home media macropad, my use case is fine.

SO EXCITED TO HAVE THIS RUNNING!

closed 2025-03-25 by chrisallick