T-vK/ESP32-BLE-Keyboard #364: Unreliable sending above 18 characters

ChrisXenon, 2025-05-25:
I'm running on an ESP32 Dev board connected by BT to a Windpws 11 PC being fed characters into the app Notepad++ as a test.
I am programming and uploading using the Arduino IDE.
I have two buttons whose low transition is attached to an interrupt, whose ISR is:

void do_Key1() {
  detachInterrupt(Key1);
  Key1Pressed = true;
}

Loop interrogates variable Key1Pressed:

if (Key1Pressed) {
    Key1Pressed = false;
    bleKeyboard.print("123456789A12345678");
    attachInterrupt(Key1, do_Key1, FALLING);
 }

This all works fine as long as the character array being sent is less than 17 characters.
After that it becomes unreliable and the longer, the more unreliable.
What's received is either truncated and/or the last character is repeated enlessly:

123456789A1234566666666666666666666666666666

If I press the key again the repetition is stopped.

bleKeyboard.setDelay() doesn't help.

Any thoughts?

ekoslav, 2025-05-25:
I have run into this issue a long time ago. My solution to this problem was to split it to individual characters and send as a stream instead of a group.

Here is my example:

`
void SendSequence(String sequence){
  // send sequence to bluetooth one char at the time
  Serial.print("[INFO]: typing speed is ");
  Serial.print(typerate);
  Serial.println(" ms");
  Serial.println(sequence);
  int len = sequence.length();
  Serial.println(len);
  //Serial.println("---------------");
  int special=0;
  for (int i=0;i<=len;i++) {
    // analysis of the string
    //is it special char?
      if (special == 1) {
            if (sequence.charAt(i) == 'r') {special=0;bleKeyboard.write(KEY_RETURN);Serial.println("[KEY_RETURN]");}    //add CR
            else if (sequence.charAt(i) == 's') {special=0;bleKeyboard.write(32);Serial.println("[SPACE]");}    //add space
            else if (sequence.charAt(i) == 'c') {special=0;bleKeyboard.write(':');Serial.println(":");}    //add colon
            else if (sequence.charAt(i) == 't') {special=0;bleKeyboard.write(KEY_TAB);Serial.println("[KEY_TAB]");}       //add horizontal tab
            else if (sequence.charAt(i) == '\\') {special=1;bleKeyboard.print('\\');Serial.println("[\\]");}       //Dual slash situation
            else {
                //if not expected special char,turn off special, and add it to the string
                special=0;
                //bleKeyboard.print("\\");
                //bleKeyboard.print(sequence.charAt(i));
                }                        
          }        
      else { //regular char
        if(sequence.charAt(i) == '\\') {
          special=1;
          } //escape char, turn on special for next char
        else {
          //bleKeyboard.print(sequence.charAt(i));
        }
      // to avoid repeating characters
      //bleKeyboard.releaseAll();
      //Serial.println();
      delay(typerate);
      }   
    }
  Serial.println("---------------");
  }
`

ChrisXenon, 2025-05-26:
Ekoslav, thanks a lot for your help. Sending by character was my next step, but it's good to know you found the same problem and that thiss solved it, and - yeah - I'll rip that code, too. Interesting about the random delays also - makes a kind of sense.

I guess the library author might fix the issue at some point.

Thanks again,
Chris

ChrisXenon, 2025-05-26:
Ekoslav, if I am ask - why do you add space and colon as special characters - why can't I just send them natively, with  bleKeyboard.print(" ");, for example?

Chris