T-vK/ESP32-BLE-Keyboard #368: Error after enabling NimBLE for ESP32 (platformio)

javierrayhan, 2025-07-09:
Hi there,

I’m currently working on a macro keyboard project using ESP32 and the [ESP32 BLE Keyboard](https://github.com/T-vK/ESP32-BLE-Keyboard) library. Everything was going fine until I realized the default BLE stack consumes too much flash and memory — so I decided to switch to NimBLE for its lighter footprint.

However, after enabling #define USE_NIMBLE, the project fails to compile with the following errors:

In file included from src/BLEKeyboardWrapper.h:4,
                 from src/main.cpp:3:
.pio/libdeps/esp32dev/ESP32 BLE Keyboard/BleKeyboard.h:2: warning: "USE_NIMBLE" redefined
 #define USE_NIMBLE

<command-line>: note: this is the location of the previous definition
In file included from src/BLEKeyboardWrapper.h:4,
                 from src/BLEKeyboardWrapper.cpp:1:
.pio/libdeps/esp32dev/ESP32 BLE Keyboard/BleKeyboard.h:131:60: error: expected class-name before ',' token
 class BleKeyboard : public Print, public BLEServerCallbacks, public BLECharacteristicCallbacks
                                                            ^
In file included from src/BLEKeyboardWrapper.h:4,
                 from src/BLEKeyboardWrapper.cpp:1:
.pio/libdeps/esp32dev/ESP32 BLE Keyboard/BleKeyboard.h:19:36: error: 'NimBLEAdvertising' does not name a type; did you mean 'BLEAdvertising'?
 #define BLEAdvertising             NimBLEAdvertising
                                    ^~~~~~~~~~~~~~~~~
.pio/libdeps/esp32dev/ESP32 BLE Keyboard/BleKeyboard.h:138:3: note: in expansion of macro 'BLEAdvertising'
   BLEAdvertising*    advertising;
   ^~~~~~~~~~~~~~
In file included from src/BLEKeyboardWrapper.h:4,
                 from src/BLEKeyboardWrapper.cpp:1:
.pio/libdeps/esp32dev/ESP32 BLE Keyboard/BleKeyboard.h:176:16: error: 'virtual void BleKeyboard::onConnect(NimBLEServer*)' marked 'override', but does not override        
   virtual void onConnect(BLEServer* pServer) override;
                ^~~~~~~~~
.pio/libdeps/esp32dev/ESP32 BLE Keyboard/BleKeyboard.h:177:16: error: 'virtual void BleKeyboard::onDisconnect(NimBLEServer*)' marked 'override', but does not override
   virtual void onDisconnect(BLEServer* pServer) override;
                ^~~~~~~~~~~~
.pio/libdeps/esp32dev/ESP32 BLE Keyboard/BleKeyboard.h:178:16: error: 'virtual void BleKeyboard::onWrite(NimBLECharacteristic*)' marked 'override', but does not override  
   virtual void onWrite(BLECharacteristic* me) override;
                ^~~~~~~
*** [.pio\build\esp32dev\src\BLEKeyboardWrapper.cpp.o] Error 1
In file included from src/BLEKeyboardWrapper.h:4,
                 from src/main.cpp:3:
.pio/libdeps/esp32dev/ESP32 BLE Keyboard/BleKeyboard.h:131:60: error: expected class-name before ',' token
 class BleKeyboard : public Print, public BLEServerCallbacks, public BLECharacteristicCallbacks
                                                            ^
In file included from src/BLEKeyboardWrapper.h:4,
                 from src/main.cpp:3:
.pio/libdeps/esp32dev/ESP32 BLE Keyboard/BleKeyboard.h:19:36: error: 'NimBLEAdvertising' does not name a type; did you mean 'BLEAdvertising'?
 #define BLEAdvertising             NimBLEAdvertising
                                    ^~~~~~~~~~~~~~~~~
.pio/libdeps/esp32dev/ESP32 BLE Keyboard/BleKeyboard.h:138:3: note: in expansion of macro 'BLEAdvertising'
   BLEAdvertising*    advertising;
   ^~~~~~~~~~~~~~
In file included from src/BLEKeyboardWrapper.h:4,
                 from src/main.cpp:3:
.pio/libdeps/esp32dev/ESP32 BLE Keyboard/BleKeyboard.h:176:16: error: 'virtual void BleKeyboard::onConnect(NimBLEServer*)' marked 'override', but does not override        
   virtual void onConnect(BLEServer* pServer) override;
                ^~~~~~~~~
.pio/libdeps/esp32dev/ESP32 BLE Keyboard/BleKeyboard.h:177:16: error: 'virtual void BleKeyboard::onDisconnect(NimBLEServer*)' marked 'override', but does not override     
   virtual void onDisconnect(BLEServer* pServer) override;
                ^~~~~~~~~~~~
.pio/libdeps/esp32dev/ESP32 BLE Keyboard/BleKeyboard.h:178:16: error: 'virtual void BleKeyboard::onWrite(NimBLECharacteristic*)' marked 'override', but does not override  
   virtual void onWrite(BLECharacteristic* me) override;
                ^~~~~~~
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1

What I’ve Tried:

Removed local #define USE_NIMBLE and used only build_flags = -DUSE_NIMBLE
Ensured NimBLEDevice.h is included
Confirmed class aliases (#define BLEServer NimBLEServer, etc.) are being used
Still, the virtual methods using BLEServer* are not overriding properly when NimBLE is enabled.

Thanks in advance!

webmonkey, 2025-07-19:
This workaround seems to exist.

> I've temporality worked around the issue by using the old version of NimBLE in PlatformIO like so:
> 
> ```
> 
> lib_deps = 
> 
>     T-vk/ESP32 BLE Keyboard@^0.3.2-beta
> 
>     NimBLE-Arduino@^1.4.3
> 
> build_flags = 
> 
>     -D USE_NIMBLE
> 
> ``` 

 _Originally posted by @byerss in [#340](https://github.com/T-vK/ESP32-BLE-Keyboard/issues/340#issuecomment-2568742259)_

javierrayhan, 2025-07-22:
Thanks a lot for sharing this workaround!, I haven't tried it yet with platformIO, i'll give ait a shot soon. Appreciate the tip!

JuniorPolegato, 2025-07-31:
Hi!

I made some adjustments for compatibility with new versions of BLE and NimBLE: https://github.com/T-vK/ESP32-BLE-Keyboard/pull/372

javierrayhan, 2025-10-03:
Thank you for all help about this issue, i will try it ASAP.